nefarius/ViGEmBus #141: Publish to Windows Package Manager (WinGet)

vedantmgoyal2009, 2022-07-25:
**Is your feature request related to a problem? Please describe.**
N/A

**Describe the solution you'd like**
With every new release of ViGEmBus, use a GitHub workflow to update manifests at https://github.com/microsoft/winget-pkgs repository.

I can create a pull request if you would like. It would help in keeping manifests up-to-date at WinGet with every new release automatically.

Same can be done for HidHide.

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

vedantmgoyal2009, 2022-08-02:
* A PR on HidHide has been created which adds winget-releaser workflow: https://github.com/ViGEm/HidHide/pull/80

nefarius, 2022-08-13:
I haven't forgotten about those, thank you kindly, just trying to get the overall pile smaller before I look into this new universe.

gerardog, 2022-10-03:
I just upgraded ViGemBus with Winget, and my machine inadvertedly restarted, no warning, no way to cancel.

```
2022-10-03 09:40:08.712 [CORE] WinGet, version [1.3.0], activity [{A6E3A63F-2EAA-4B92-A89F-D09C962219A8}]
2022-10-03 09:40:08.713 [CORE] OS: Windows.Desktop v10.0.22000.1042
2022-10-03 09:40:08.713 [CORE] Command line Args: C:\Users\gerar\AppData\Local\Programs\WingetUI\winget-cli\winget.exe  install -e --id ViGEm.ViGEmBus --source winget --accept-source-agreements --silent
2022-10-03 09:40:08.713 [CORE] IsCOMCall:0; Caller: winget-cli
2022-10-03 09:40:08.714 [CLI ] WinGet invoked with arguments: 'install' '-e' '--id' 'ViGEm.ViGEmBus' '--source' 'winget' '--accept-source-agreements' '--silent'
2022-10-03 09:40:08.714 [CLI ] Found subcommand: install
2022-10-03 09:40:08.714 [CLI ] Leaf command to execute: root:install
2022-10-03 09:40:08.715 [CLI ] Executing command: install
2022-10-03 09:40:08.716 [REPO] Named source requested, found: winget
2022-10-03 09:40:08.718 [CORE] Started trust validation of msix at: C:\Users\gerar\AppData\Local\Microsoft/WinGet\State\defaultState\Microsoft.PreIndexed.Package\Microsoft.Winget.Source_8wekyb3d8bbwe\source.msix
2022-10-03 09:40:08.795 [CORE] Result for certificate chain validation of Microsoft origin: 0
2022-10-03 09:40:08.861 [CORE] Result for trust info validation of the msix: 0
2022-10-03 09:40:09.176 [REPO] Opening SQLite Index for ImmutableRead at 'C:\Users\gerar\AppData\Local\Temp\WinGet\defaultState\{CD097CBE-DBC4-42D2-97D7-50967E29A741}'
2022-10-03 09:40:09.176 [SQL ] Opening SQLite connection: 'file:/C:/Users/gerar/AppData/Local/Temp/WinGet/defaultState/{CD097CBE-DBC4-42D2-97D7-50967E29A741}?immutable=1' [1, 40]
2022-10-03 09:40:09.181 [REPO] Opened SQLite Index with version [1.5], last write [2022-10-03 09:23:21.000]
2022-10-03 09:40:09.238 [CLI ] Found one app. App id: ViGEm.ViGEmBus App name: ViGEm Bus Driver
2022-10-03 09:40:09.239 [REPO] Downloading manifest
2022-10-03 09:40:09.239 [CORE] WinINet downloading from url: https://cdn.winget.microsoft.com/cache/manifests/v/ViGEm/ViGEmBus/1.21.442/f9f6-ViGEm.ViGEmBus.yaml
2022-10-03 09:40:10.136 [CORE] Download hash: e3766539420a34462b156d4f361e64015ebfcc1034500370110fbb9d73988f3e
2022-10-03 09:40:10.136 [CORE] Download completed.
2022-10-03 09:40:10.137 [CLI ] Manifest fields: Name [ViGEm Bus Driver], Version [1.21.442]
2022-10-03 09:40:10.139 [CLI ] Starting installer selection.
2022-10-03 09:40:10.139 [CLI ] Completed installer selection.
2022-10-03 09:40:10.145 [CLI ] Generated temp download path: C:\Users\gerar\AppData\Local\Temp\WinGet\defaultState\ViGEm.ViGEmBus.1.21.442\155c50f1eec07bdc28d2f61a3e3c2c6c132fee7328412de224695f89143316bc
2022-10-03 09:40:10.145 [CORE] Downloading to path: C:\Users\gerar\AppData\Local\Temp\WinGet\defaultState\ViGEm.ViGEmBus.1.21.442\155c50f1eec07bdc28d2f61a3e3c2c6c132fee7328412de224695f89143316bc
2022-10-03 09:40:10.145 [CORE] DeliveryOptimization downloading from url: https://github.com/ViGEm/ViGEmBus/releases/download/v1.21.442.0/ViGEmBus_1.21.442_x64_x86_arm64.exe
2022-10-03 09:40:13.755 [CORE] Download completed.
2022-10-03 09:40:15.601 [CORE] Started applying motw to C:\Users\gerar\AppData\Local\Temp\WinGet\defaultState\ViGEm.ViGEmBus.1.21.442\155c50f1eec07bdc28d2f61a3e3c2c6c132fee7328412de224695f89143316bc with zone: 3
2022-10-03 09:40:15.605 [CORE] Finished applying motw
2022-10-03 09:40:15.605 [CLI ] Installer hash verified
2022-10-03 09:40:15.605 [CORE] Started applying motw to C:\Users\gerar\AppData\Local\Temp\WinGet\defaultState\ViGEm.ViGEmBus.1.21.442\155c50f1eec07bdc28d2f61a3e3c2c6c132fee7328412de224695f89143316bc with zone: 2
2022-10-03 09:40:15.606 [CORE] Finished applying motw
2022-10-03 09:40:15.607 [CLI ] Successfully renamed downloaded installer. Path: C:\Users\gerar\AppData\Local\Temp\WinGet\defaultState\ViGEm.ViGEmBus.1.21.442\ViGEmBus_1.21.442_x64_x86_arm64.exe
2022-10-03 09:40:15.609 [REPO] Creating PredefinedInstalledSource with filter [ARP]
2022-10-03 09:40:15.609 [REPO] Creating new SQLite Index [4294967295.4294967295] at ':memory:'
2022-10-03 09:40:15.609 [SQL ] Opening SQLite connection: ':memory:' [6, 0]
2022-10-03 09:40:15.652 [REPO] Examining ARP entries for Machine | X64
2022-10-03 09:40:15.692 [REPO] Examining ARP entries for Machine | X86
2022-10-03 09:40:15.750 [REPO] Examining ARP entries for User | X64
2022-10-03 09:40:16.002 [CLI ] Installer args: /exenoui /qn /norestart
2022-10-03 09:40:16.002 [CLI ] Starting: 'C:\Users\gerar\AppData\Local\Temp\WinGet\defaultState\ViGEm.ViGEmBus.1.21.442\ViGEmBus_1.21.442_x64_x86_arm64.exe' with arguments '/exenoui /qn /norestart'
```

I wonder if `ViGEmBus_1.21.442_x64_x86_arm64.exe` is properly forwarding `/norestart` to msiexec.

nefarius, 2022-10-03:
I removed the restart required from the most recent setup though 🤔

gerardog, 2022-10-03:
Happened a similar thing in my project gerardog/gsudo#167. In that case, a file was in use, so `MSIEXEC` (when used with /qn silent switch) would decide that restart automatically was a good idea.

Here also may be that the driver was in use in my system. I had DS4 running, and yesterday we had some steam/epic games activity, and not rebooted since. Using a PS4 controller.

nefarius, 2022-10-03:
Well, I can't really do both without things going horribly wrong, Windows does its own reboot required detection so whenever the driver is in use, no matter what I do, setup might fail if I depend the setup on a success check.

gerardog, 2022-10-03:
I managed to intercept the msiexec call

`"C:\WINDOWS\system32\msiexec.exe" /i "C:\ProgramData\Nefarius Software Solutions\ViGEm Bus Driver 1.21.442\install\5215C05\ViGEmBus.x64.msi" /qn /norestart AI_SETUPEXEPATH=C:\Users\gerar\AppData\Local\Temp\WinGet\ViGEm.ViGEmBus.1.21.442\ViGEmBus_1.21.442_x64_x86_arm64.exe SETUPEXEDIR=C:\Users\gerar\AppData\Local\Temp\WinGet\ViGEm.ViGEmBus.1.21.442\ EXE_CMD_LINE="/exenoui  /exenoupdates  /forcecleanup  /wintime 1664800881 /qn /norestart "`

So, it is forwarding the /norestart after all.

gerardog, 2022-10-03:
> Well, I can't really do both without things going horribly wrong, Windows does its own reboot required detection so whenever the driver is in use, no matter what I do, setup might fail if I depend the setup on a success check.

The problem here is `winget install /silent` often ends in sudden restarts.  Typically the /norestart fixes the problem, by leaving up to the user the choice to restart later... But in this case, I don't know.

Masamune3210, 2022-10-03:
There is a open pull request and work currently being dome behind the scenes to handle restarts better in winget

PR #142 (2022-10-24): Add Winget Releaser workflow
This action automatically generates manifests for WinGet Community Repository ([microsoft/winget-pkgs](https://github.com/microsoft/winget-pkgs)) and submits them. This is in line with the PR in HidHide (https://github.com/ViGEm/HidHide/pull/80).

Before merging this:
- Follow the steps in https://github.com/ViGEm/HidHide/pull/80. See that PR for the rest of the information.

Resolves #141
closed 2022-10-24 by nefarius