nefarius/ViGEmBus #148: Device subtypes like guitars?

joao678, 2022-08-20:
Does ViGEmBus supports different xbox device types?
I've been trying different setups with different devices and different programming languages to no avail so I was wondering if we can do something like x360ce does with 'Device Types' like connecting a virtual guitar hero guitar or maybe a steering wheel?

The closest I've got while testing was informing a VID and PID using ViGEm.NET while using the ViGEmClient.CreateXbox360Controller() function but I don't think this counts because I believe a game like Guitar Hero 3 for PC reads the device subtype specifically (and maybe older racing games use this method aswell)

**Describe the solution you'd like**
Maybe a way to inform a device type on a function like CreateXbox360Controller so that we have full control over the device created

**Describe alternatives you've considered**
For my personal project I can always emulate the keyboard with frets and the strumbar but I mean it's not the same thing, right? 🧐

**Additional context**
This might be useful: [https://docs.microsoft.com/en-us/windows/win32/xinput/xinput-and-controller-subtypes](https://docs.microsoft.com/en-us/windows/win32/xinput/xinput-and-controller-subtypes)

nefarius, 2022-08-20:
It does not, its successor will though!

closed 2022-08-20 by nefarius

joao678, 2022-08-20:
Oh cool I'll look forward to that!

nefarius, 2023-03-28:
Supported in v2, see #139