nefarius/ViGEmBus #152: Erratic inputs on "source" controller

ProgMetalMan, 2022-09-20:
Hello, I'm not sure if this is a bug within the scope of this project— heck, it could even be confounded by a random controller driver lying around my PC— but I thought I'd try here. 

I had this problem using BetterJoy (when it worked at all...) and DSAdvance, both of which use ViGEmBus. Basically, I'm trying to use a Switch Pro controller wrapped in an X360 emulator. Where the "Pro Controller" entry on Windows' Gamepad list would normally be entirely unresponsive, now shows a constant sporadic stream of inputs. I think some kind of datastream is spuriously being interpreted as gamepad inputs, but I have no clue how that can actually happen. 

To reproduce, simply use a ViGEmBus-based controller wrapper and plug in a Switch Pro controller. If the wrapper works correctly, the emulated controller will successfully respond to the inputs on the switch controller. But then, look at the original entry on the gamepad list. At this point, I see sporadic inputs on all buttons and axes, where I would expect to see no input, as I had before invoking the wrapper.

Brief clip demonstrating the sporadic inputs: https://cdn.discordapp.com/attachments/746230692039622696/1021169036593217617/fresh_hell.mp4. 

System details (please complete the following information):
 - OS: Windows 10 Pro 21H2
 - Feeder software: Consistent behavior with both BetterJoy for Windows v7.1 as well as DSAdvance 0.6.2
 - Driver Version: 1.21.442_x64_x86_arm64

I've used Switch Pro controllers on my system before, but I started to notice this problem when trying to configure inputs for MAME. The inputs on the "2nd controller" cause the UI to go haywire.

Kanuan, 2022-09-20:
so, let's go in parts.

# 1 

> I've used Switch Pro controllers on my system before, but I started to notice this problem when trying to configure inputs for MAME. The inputs on the "2nd controller" cause the UI to go haywire.

This is usually called "double input" or "double controller" issue.

You need to HIDE your REAL controller (Pro Controller) so games/apps can only pick its Virtual counterpart (X360). The driver that allows you to do this is called HidHide. For more info: [HidHide Simple Setup Guide](https://vigem.org/projects/HidHide/Simple-Setup-Guide/).

Basically:

1. Hide your real controller once via USB and once via BT if you also use it wirelessly
    - HidHide assumes your device via BT and USB are different, so you need to hide it twice
1. Allow the remapper of your choice to see hidden devices, otherwise it's like you have no controller connected at all

---------------------

# 2

> Where the "Pro Controller" entry on Windows' Gamepad list would normally be entirely unresponsive, now shows a constant sporadic stream of inputs

I don't have much knowledge with how the pro controller operates but my guess is that:

- Your remapper is sending a request for the controller to change to some special mode
- In this special mode, it communicates differently, but Windows still tries to "read" it as per the previous way. Because how Windows tries to read the new info does not match what it expects things go crazy
- If your remapper is correctly reading the controller anyway than it doesn't make much difference since you'll hide the real controller anyway

----------

# 3 

Ryochan7's DS4Windows has support for Pro Controllers in case you doesn't know. Just in case you want to test other remappers.

Ryochan7, 2022-09-20:
Not valid. Even Steam Input makes the exposed DirectInput interface for the Switch Pro (connected via USB) behave like this. Not exactly sure why the mode switch causes this but it is expected behavior that has nothing to do with ViGEmBus. You need to use HidHide to hide the Switch Pro controller from other apps on your system and only expose the emulated controller created with ViGEmBus.

ProgMetalMan, 2022-09-20:
Understood. Thank you for the help!

closed 2022-09-20 by ProgMetalMan