nefarius/ViGEmBus #153: [Enhancement] Remove the Windows Server restrictions from the ViGEmBus installer

akemin-dayo, 2022-09-28:
The ViGEmBus 1.21.442 installer explicitly blocks installation on all Windows Server OSes, presumably due to the fact that the XInput controller drivers are not installed by default on Windows Server.

However, this occurs even if the system in question has those drivers installed.

It would be nice if these restrictions were removed (as was the case in previous ViGEmBus installers), so the somewhat-hacky below workaround would no longer be required.

## Workaround
1. Extract the ViGEmBus installer EXE by running `./ViGEmBus_1.21.442_x64_x86_arm64.exe /extract vigembus`.
2. Inside the extracted folder (which may end up being a random string like `5215C05`), right click on `ViGEmBus.inf` and click "Install".
	* **※ NOTE:** If your system architecture is _not_ x86_64, look inside the "32-bit" or "ARM64" folders instead.
3. Reboot your system.
4. Verify that the new version of "Nefarius Virtual Gamepad Emulation Bus" appears in Device Manager, under "System Devices".

nefarius, 2022-09-28:
I will not remove that restriction because I made the setup for individuals, gamers on their home machine running a client OS, not people who use it on cloud machines or in Enterprise setups without ever contributing anything back to the project 🙂

nefarius, 2022-09-28:
> Ah, so it's intended behaviour. Well, that's fair.

Yes, I'm afraid you suffer the consequences from past abuse of my craft, sorry for the inconveniences but I'll stick to my principles.

closed 2022-09-28 by nefarius

akemin-dayo, 2022-09-28:
Mnn, sorry to hear about that :/

In any case, it's just a minor inconvenience anyway (though I am… a developer, so not exactly a typical user ;P)

I _do_ mainly use ViGEmBus (+ DS4Windows) on my local Windows 10 machine, I just also happen to have a Google GCE VM that I play around with occasionally since I was given some free credits that have an expiration date ;P

No worries though, it's your project. Really happy to see the new features recently!

Anopetia, 2022-10-09:
@akemin-dayo  Thank you so much for this help! Life saver :)

Astrian, 2022-10-14:
Thanks, @akemin-dayo, and I understand and appreciate the work by @nefarius. However, is it necessary that put a tip inside the installer to explain this? I was struggling with this problem for two days 😂

nefarius, 2022-10-15:
I updated the README accordingly. The installer is a 3rd party product, not sure it that is possible, I'll put it on the pile of things to eventually look into 😛

Nanorulez123, 2023-03-28:
Some home users use server 2022 for the higher uptime ect for 24/7 usage as a game server, most corps would use linux anyway

Masamune3210, 2023-03-28:
Not gonna happen. Not sure why you thought commenting on another issue after being told no in the one you made would end any better

Masamune3210, 2023-03-29:
As for the argument about uptime, this is my computer running Windows 11 Pro
[image]

No hibernate, no reboot, no sleep. You don't need server to have long stable uptimes, you need a not broken to pieces install