nefarius/ViGEmBus #154: Can't install with cloud drive mounted

mixtern, 2022-10-04:
I'm using a program called Disk-O to access my cloud files through file explorer. The cloud is mounted to the letter `O:`
When I got an update message for ViGEmBus and tried installing it, I get the 'Invalid Drive O' error

[image]

Just for clarity, I'm not trying to install ViGEm to drive O, I'm using the default install path - 'C:\Program Files\Nefarius Software Solutions\ViGEm Bus Driver\'. When I turn Disk-O off, the install goes the intended way, and there are no problems with the program itself after turning Disk-O back on.

Here are the logs:
[install.log](https://github.com/ViGEm/ViGEmBus/files/9702558/install.log)
[setupapi.dev.log](https://github.com/ViGEm/ViGEmBus/files/9702559/setupapi.dev.log)

Masamune3210, 2022-10-04:
its a msiexec bug, same things happens with rclone mounts

nefarius, 2022-10-04:
What @Masamune3210 said, out of my power to fix, sorry.

closed 2022-10-04 by nefarius