nefarius/ViGEmBus #159: Does not detect as a Xbox 360 controller in Windows 11

itsalexdc, 2022-10-15:
**Describe the bug**
Does not detect as a Xbox 360 controller in Windows 11

**To Reproduce**
Just try to use Game pass games

**Expected behavior**
For it to work with Game pass games on Windows 11

**Screenshots**
[image]

**System details (please complete the following information):**
 - OS: Windows 11 Home Single Language 21H2
 - Feeder software: SWICD v.0.3.0-0dc3c0b-release
 - Driver Version: 1.21.442_x64_x86_arm64

**Additional context**
For Steam Deck

Masamune3210, 2022-10-15:
Can verify a legitimate ds4, dualsense, and Xbox one controller all work perfectly fine with DS4Windows at least, sounds like a issue with whatever software you are using to handle the controller output

nefarius, 2022-10-15:
What @Masamune3210 said 👍

closed 2022-10-15 by nefarius

itsalexdc, 2022-10-15:
but why would you use this software with an xbox controller? This is related to Steam Deck xbox 360 controller emulation.

Masamune3210, 2022-10-15:
The point isn't what software is consuming it this point is that ViGEM is working correctly

Masamune3210, 2022-10-15:
The interface doesn't do any conversion by itself it's only purpose is to consume physical inputs and offer them as something that another program can intake and output as it wants

itsalexdc, 2022-10-15:
Oh I see. My apologies. Thank you for responding.

Masamune3210, 2022-10-15:
No problem