nefarius/ViGEmBus #163: DualSense Support?

anar4732, 2022-12-17:


Masamune3210, 2022-12-17:
The library isnt involved with this. All it does is take input from one thing and convert it to another thing, its up to the system or a tool like DS4Windows to handle getting things connected if needed and using the library to do whatever it wants to do with the data

nefarius, 2022-12-17:
> The library isnt involved with this. All it does is take input from one thing and convert it to another thing, its up to the system or a tool like DS4Windows to handle getting things connected if needed and using the library to do whatever it wants to do with the data

I think OP is asking about DualSense **emulation** support which is a valid question and on the todo for quite some time.

nefarius, 2022-12-17:
Closing, see #139

closed 2022-12-17 by nefarius