nefarius/ViGEmBus #164: Is the ds4 bt audio still worked on?

Pro55624, 2023-01-18:
Ryochan made some impressive progress but its only updates on this are well not in any way affecting the actual feature. He kinda gave up since he didnt have enough support and time to figure it out.

nefarius, 2023-01-18:
It is but it's reached a dead end that is difficult to explain since the source of the problem is not yet well understood. Is it an implementation issue? Or OS shipped driver issue? Nobody knows, nobody helps, so it depends on reverse engineering, trial and error and last but not least: luck.

nefarius, 2023-01-24:
Even if, what good would that be for? I can't and won't delegate work to random hobbyists... That's not how open-source works.