nefarius/ViGEmBus #172: PS4 Remote Play Controller emulation stopped working

Walter-o, 2023-03-20:
Virtual controllers created with ViGEm are currently not working
due to an update in PS4 Remote play software

Alleged reason is a special check being added in a software update but i personally can't confirm this.

Another earlier Github issue at DS4Windows talks about the same thing:
https://github.com/Ryochan7/DS4Windows/issues/2868

This bug is replicable without a special setup, i couldn't get past it with 5+ different methods.

I'm currently in search for a workaround aa

nefarius, 2023-03-20:
Can confirm, v6 of the software is exchanging additional control request packets which is not done by ViGEm in contrast to the real hardware.

Walter-o, 2023-03-30:
I found a sort of workaround for what i was trying to accomplish.
By using "Chiaki" as remote play program i was able to control without a genuine DS4 controller again.

I now have to pray that sony won't break third-party remoteplay apps now. 🙏

Masamune3210, 2023-03-31:
They really can't in a way that also wouldn't break it for ps4's as well without a update, due to the way chiaki works